mrbeam/MrBeamPlugin #3: Shutdown/Reboot button is not showing up

make-ing, 2016-08-03:
A user mentioned that the "Shutdown/Reboot" button is not showing up anymore.

make-ing, 2016-08-03:
Clemens can you check if this is true (a bug)? And if so fix it :)

hungerpirat, 2016-08-03:
could be a config file / OctoPi related thing

make-ing, 2016-08-04:
it is indeed a config file problem. You need to add the following to the config.yaml file:

```
system:
  action
  - name: Shutdown
    action: shutdown
    command: sudo shutdown -h now
    confirm: You are about to shutdown the system.
  - name: Reboot
    action: reboot
    command: sudo shutdown -r now
    confirm: You are about to reboot the system.
```

make-ing, 2017-01-30:
fixed in mrbeam variant of OctoPi build: [commit](https://github.com/make-ing/OctoPi/commit/916d378f28df99fbd3b01afb3fe0db5f2f79f3b5)

closed 2017-01-30 by make-ing