mrbeam/MrBeamPlugin #43: Integrate DXF import plugin from Nick

hungerpirat, 2017-04-03:
Steps to do: 

- fork the plugin
- integrate js
https://github.com/mrbeam/MrBeamPlugin/blob/master/octoprint_mrbeam/__init__.py#L101

- enable file type dxf
https://github.com/mrbeam/MrBeamPlugin/blob/8c159d3006cf6ff9c21ec47dac50fcb1790dc8dd/octoprint_mrbeam/__init__.py#L613

- knockout view template for dxf (if necessary)
https://github.com/mrbeam/MrBeamPlugin/blob/8c159d3006cf6ff9c21ec47dac50fcb1790dc8dd/octoprint_mrbeam/static/js/working_area.js#L616

- implement placeDXF() function analog to placeSVG() in working_area.js

Gallore, 2017-04-13:
Get a DXF file from thingiverse.com (search for lasercut)

hungerpirat, 2017-05-17:
DXF support integrated.

commit a487b244 (2017-05-17): closes #43
closed 2017-05-17 by hungerpirat