mrbeam/MrBeamPlugin #47: Mixpanel issue: #printer_conncet button -> redirect to /undefined -> 404

Gallore, 2017-04-25:
With Mixpanel activated, a click on the #printer_connect (Connect Button), the browser gets directed to /undefined by some mixpanel.js which of course leads to an 404.

Tried: 
- add a href="#" to the button
- added a custom callback to the mixpanel.track_links call this element. It gets called but the problematic thing happens much later....

Disabled Mixpanel by default for now. (Add "analyticsEnabled: true" to mrbeam section in config.yaml to enable.)

Gallore, 2017-04-25:
Mispanel JS documentation: https://mixpanel.com/help/reference/javascript

yvan33, 2017-11-07:
@Gallore - I experience the same bug, thanks for reporting it. I think it's critical because it prevents people to use `track_links`.

@hungerpirat - Any idea when a fix will go in?

Gallore, 2017-11-08:
@yvan33 : Not sure what you mean. Are you aware that this repo is about out Mr Beam Lasercutter software? This is NOT a Mixpanel repo.

yvan33, 2017-11-08:
@Gallore - Right, I realized that was not be the best place to post it.

But I believe to fix your bug you'll need to change this line: https://github.com/mrbeam/MrBeamPlugin/blob/215b94b570dcfb81e0d8926c43be49d49e7daf93/octoprint_mrbeam/templates/mrbeam_ui_index.jinja2#L185
to be `mixpanel.track_links(element, 'link '+ href, {}, () => false);`

The same probably apply to l179. Mixpanel documentation is not up to date regarding the 4th parameter of `track_links`

Gallore, 2017-11-13:
@yvan33 : Thank you :-)

hungerpirat, 2018-08-29:
kicked out mixpanel.

commit 2da30b99 (2018-08-30): fixes #47
closed 2018-08-29 by hungerpirat