mrbeam/MrBeamPlugin #65: Machine profile extensions

hungerpirat, 2017-05-09:
The machine profile should be extended with the following settings:

connection (port, baudrate, autoconnect)

presence of hardware features like
- temperature
- exhaust system (filter or fan only, control)
- camera
- one button

camera calibration parameters

laser source parameters
- power
- wavelength
- dimmable

gcode style / gcodes for
- request status
- trigger connection response
- before/after path
- before/after job
(- before/after image)
- image overshoot distance
- after homing code
- setting laser power
- setting speed
- G2/G3 support

additionally the grbl settings should be "synced":
- homing direction
- axes direction
- working area size + homing pulloff

required safety warnings 
- glasses ?

focus howto

... more suggestions are welcome.

hungerpirat, 2017-05-09:
+gantry style: cartesian, HBot, coreXY

Gallore, 2017-05-10:
Implementation of how profiles are read/stored and default values are defined needs to be redone. Current implementation old /  buggy / unreliable / I don't understand it. (Not exactly sure which one.)

Gallore, 2017-05-10:
current lasercutter profiles are in 
.octoprint/plugins/ 
but they should be in 
.octoprint/data/mrbeam/