mrbeam/MrBeamPlugin #68: gradients are ignored for engraving

hungerpirat, 2017-05-10:
the gradient definition is probably refering to the defs tag, which is somehow lost when generating the raster image.

'''
<path id="rect2985" fill="url('#linearGradient2993')" stroke="#000000" style="color:#000000;display:inline;overflow:visible;visibility:visible;fill:url(#linearGradient2993);fill-opacity:1;fill-rule:nonzero;stroke:#000000;stroke-width:0.5;marker:none;enable-background:accumulate" transform="matrix(1,0,0,1,0,0)" d="M17.7692,961.3368800000001L433.11005,961.3368800000001L433.11005,996.769951L17.7692,996.769951L17.7692,961.3368800000001Z" mb:color="#000000"></path>
'''

hungerpirat, 2017-05-10:
Little investigation together with @Gallore : 
Snap can't preserve ids while cloning.
This snippet might help out:

```javascript
    Snap.plugin( function( Snap, Element, Paper, global ) {
      Element.prototype.cloneKeepIds = function() {
        var cloneNode = this.node.cloneNode(true);
        var cloneElt  = new Element(cloneNode);
        cloneElt.insertAfter(this);
        return cloneElt;
      };
    });
```

original file:
[engraving_test_pattern.svg.zip](https://github.com/mrbeam/MrBeamPlugin/files/989310/engraving_test_pattern.svg.zip)

Gallore, 2017-10-04:
@hungerpirat : is this still relevant?

make-ing, 2017-11-17:
still relevant!!!

hungerpirat, 2020-01-19:
in the meantime id preserving was fixed, but coordinate issues appear. => duplicate of #426

closed 2020-01-19 by hungerpirat