mrbeam/MrBeamPlugin #109: Implement unproportional scaling of user graphics in the working area.

Gallore, 2017-07-03:
It should be possible to do a unproportional scaling of graphics on the working area. However, this should not be the default action. 

We need a usability concept for this one: We could do the unproportional scaling while the user presses a modifier key (eg: strg). But this would be in contrast to most graphic editors were you need to press the modifier to HAVE the proportional scaling.

irlaec, 2020-10-12:
[image] [Mirrorring + Proportional scaling](https://trello.com/c/b8AFPcG3/682-mirrorring-proportional-scaling)

hungerpirat, 2021-06-28:
Nice one. Reported 2017 -> fixed recently, already released. @Gallore

closed 2021-06-28 by hungerpirat