mrbeam/MrBeamPlugin #122: Bug: OctoPrint restart doesn't trigger internal shutdown routines.

Gallore, 2017-07-17:
After a software update the regular shutdown routine is not executed... Looks like the process is just getting killed after some time. (Not sure if this is only in case of an software update.)

Andy: 

> Und bei nen Restart zB durch n Software Update. Da gibt es keinen Shutdown event und on_shutdown() wird auch nciht aufgerufen, richtig?

Gina Häußge (foosel@gmail.com):

> Kommt drauf an. Bei einem Update nutzt er das restart command. Das ist unter octopi sudo service octoprint restart. Sprich, init script. start-stop-daemon. Und damit sigterm gefolgt von sigkill.

A shutdown looks like this (in the logs) and a restart should look more or less the same:
```
2017-07-17 11:43:59,669 - octoprint.server - INFO - Shutting down...
2017-07-17 11:44:00,532 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2017-07-17 11:44:00,535 - octoprint.plugins.mrbeam.iobeam.lidhandler - DEBUG - onEvent() event: Shutdown, payload: None
2017-07-17 11:44:00,539 - octoprint.plugins.mrbeam.iobeam - DEBUG - shutdown() args: ('Shutdown', None)
2017-07-17 11:44:00,540 - octoprint.plugins.mrbeam - DEBUG - on_event Shutdown: None
2017-07-17 11:44:00,550 - octoprint.events - INFO - Event loop shut down
2017-07-17 11:44:00,563 - octoprint.plugins.mrbeam - INFO - on_shutdown()
2017-07-17 11:44:00,564 - octoprint.plugins.mrbeam.iobeam - DEBUG - shutdown() args: ()
2017-07-17 11:44:01,161 - octoprint.plugins.mrbeam.iobeam - DEBUG - Closing socket...
2017-07-17 11:44:01,162 - octoprint.plugins.mrbeam.iobeam - DEBUG - Worker thread stopped.
2017-07-17 11:44:02,567 - octoprint.plugins.mrbeam - DEBUG - on_shutdown() sleept
2017-07-17 11:44:02,568 - octoprint.server - INFO - Goodbye!
```

In case of a restart after a software update the log looks like this:
```
2017-07-17 11:44:22,408 - octoprint.plugins.softwareupdate - INFO - Update of rpi-ws281x to 3939c0729d51fcf085d8fb4be81700e5c9c7c4bc successful!
2017-07-17 11:44:22,667 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-07-17 11:44:22,669 - octoprint.plugins.softwareupdate - INFO - Restarting...
2017-07-17 11:44:22,752 - octoprint.plugins.mrbeam.iobeam - WARNING - Connection ended from other side. Closing connection...
2017-07-17 11:44:22,755 - octoprint.plugins.mrbeam.iobeam - DEBUG - Closing socket...
2017-07-17 11:44:22,755 - octoprint.plugins.mrbeam.iobeam - DEBUG - Sleeping for a sec before reconnecting...
2017-07-17 11:44:22,757 - octoprint.plugins.mrbeam.iobeam.onebutton_handler - DEBUG - onEvent() event:iobeam.disconnect, payload:None - shutdown_state:0 - shutdown_prepare_was_initiated_during_pause_saftey_timeout:None, is_ready_to_laser():False, ready_to_laser_ts:-1, ready_to_laser_flag:False, ready_to_laser_file:None, ready_to_laser_timer:None, print_started:-1, pause_laser_ts:-1, pause_need_to_release:False, pause_safety_timeout_timer:None, _is_during_pause_waiting_time():False, behave_cooling_state:False, _printer.get_state_id():LOCKED, _printer.is_operational():True, _iobeam.is_interlock_closed():True
2017-07-17 11:44:22,757 - octoprint.plugins.mrbeam - DEBUG - on_event iobeam.disconnect: None
2017-07-17 11:44:22,876 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.178.117
2017-07-17 11:44:22,881 - octoprint.plugins.mrbeam.iobeam.onebutton_handler - DEBUG - onEvent() event:ClientClosed, payload:{'remoteAddress': '192.168.178.117'} - shutdown_state:0 - shutdown_prepare_was_initiated_during_pause_saftey_timeout:None, is_ready_to_laser():False, ready_to_laser_ts:-1, ready_to_laser_flag:False, ready_to_laser_file:None, ready_to_laser_timer:None, print_started:-1, pause_laser_ts:-1, pause_need_to_release:False, pause_safety_timeout_timer:None, _is_during_pause_waiting_time():False, behave_cooling_state:False, _printer.get_state_id():LOCKED, _printer.is_operational():True, _iobeam.is_interlock_closed():True
2017-07-17 11:44:22,885 - octoprint.plugins.mrbeam.iobeam.onebutton_handler - DEBUG - unset_ready_to_laser()
2017-07-17 11:44:22,890 - octoprint.plugins.mrbeam - DEBUG - on_event ClientClosed: {'remoteAddress': '192.168.178.117'}
2017-07-17 11:44:23,758 - octoprint.plugins.mrbeam.iobeam - WARNING - IoBeamHandler not able to connect to socket /var/run/mrbeam_iobeam.sock, reason: [Errno 111] Connection refused. I'll keept trying but I won't log further failures.
```

amrsoll, 2019-11-29:
```
path/to/MrBeamPlugin$ grep -R on_shutdown  .
./octoprint_mrbeam/__init__.py: def on_shutdown(self):
```
Seems like the function is not used at all

amrsoll, 2020-07-02:
>Seems like the function is not used at all

Is used by OctoPrint.

Also this behaviour now seems to happen in the calibration screen, when the boardDetector is running with at least one image taken.