mrbeam/MrBeamPlugin #129: Gray Outlines in Frontend get bigger than design

clemniem, 2017-07-25:
If a design is made smaller (a lot smaller), at some point the gray outlines start to look funny and not right.

hungerpirat, 2017-07-25:
@clemniem are the grey outlines part of the design? or are you talking about the transformation bounding  boxes? 
Please provide a screenshot.

clemniem, 2017-07-25:
Yes, its the grey bounding box. See the large Mr Beam Logo on 2747 or 832F.
[image]

Gallore, 2017-10-04:
@clemniem : Is this still reproducible? Is assume it's fixed now. Please test and close if so. thx

clemniem, 2017-10-04:
It is still reproducible. Just take any large design and make it very small.

commit 89f0c605 (2020-01-17): fixes #129 and other freetransform bugs.
closed 2020-01-17 by hungerpirat