mrbeam/MrBeamPlugin #143: quicktext/gcode name error

clemniem, 2017-09-26:
If you try to engrave just a dot, the filename is '..gco' and then throws an exception, because the text engraved is the filename, this should not be the case.

clemniem, 2017-09-27:
What should happen in: convert.js line 456 _sanitize(name).

hungerpirat, 2017-09-27:
My assumption: The gcode filename is generated automatically based on the
origin filenames of the placed desigs. E.g. foo.svg => foo.gco , foo.svg +
bar.svg + bam.svg => foo_and_3more.gco
When using the quicktext-feature, there is no real filename anymore, it
uses the content as a placeholder.
the _sanitize should ensure that the generated filename is a valid
filename. This probably fails if the input is just '.'

clemniem, 2017-09-27:
yes, that is the problem. The question was more like: what kind of names should be allowed, and what to do with an invalid one?

hungerpirat, 2017-09-28:
If the name is invalid, I would suggest to generate a name with a
timestamp.

clemniem, 2017-09-28:
if an invalid filename is found, it will prepend 'mb' and append a timestamp to the .gco filename.

commit 64615e06 (2017-09-28): fixed #143 sanitize name
closed 2017-09-28 by clemniem