mrbeam/MrBeamPlugin #146: Photo button (take picture) doesnt work

clemniem, 2017-09-27:
the photo button on front page: upload/text/**photo**
No pop up asking me for permissions either. Is this a settings thing?

hungerpirat, 2017-09-28:
maybe we should update the js library. Checkout
https://github.com/mrbeam/MrBeamPlugin/blob/master/octoprint_mrbeam/static/js/lib/photobooth_min.js

clemniem, 2017-09-28:
I think we are using the latest version, but it is never called. 

There could be some issue with https and modern chrome browsers. But can't really test it right now.

Is there any logic in the jinja file thats loaded? 
https://github.com/mrbeam/MrBeamPlugin/blob/master/octoprint_mrbeam/templates/take_photo.jinja2

Does it work on someones computer?

Gallore, 2017-10-04:
See also #105

closed 2020-07-02 by amrsoll