mrbeam/MrBeamPlugin #149: terminal input is not safe

clemniem, 2017-09-27:
don't we check input on frontend, and not on backend side? 
eg. I just typed 'fan:off' and my mr_beam has an error and lost connection and is blinking red

Gallore, 2017-10-04:
Not a problem once the terminal is not accessible to regular users. #164
Let's double check the regExp that verifies the commands.

Verify that this can't be exploited for other users in the network. (API key required? Login required?)

make-ing, 2017-10-11:
there is only one regex which checks if the command is a eeprom command and does not allow it if the printer is printing.

We need to discuss if we really want to restrict terminal input or not. Since it will not be accessible to the regular user anyhow may we can leave it as is for now.

closed 2019-07-23 by Gallore