mrbeam/MrBeamPlugin #152: machine switching between "two" adresses

clemniem, 2017-09-28:
sometimes its under: http://192.168.0.10/
sometimes its under: MrBeam2-56D4.local

Is this something the mrbeam can decide or is it up to the router?

Gallore, 2017-10-04:
Technically both should work at the same time.
However: it's interesting to check why find.mrb sometimes shows one and then the other. And yes, would be nice to have the hostname have highest prio.

clemniem, 2017-10-04:
> Technically both should work at the same time.
They didn't. 'MrBeam...' did not work sometimes, only '192....'

Gallore, 2017-10-04:
OMG. This is pretty f**ked up....

Gallore, 2017-10-04:
Let's see, if we get more issues from users. For now, let's don't put effort in there.

clemniem, 2017-11-06:
I kept an eye on this issue, and if find.mr-beam.org sends me to the 192... the other one was working, so maybe it was just an issue with our router... 
Would be cool anyways if it always picks the mrbeamXXXX.local address with priority.