mrbeam/MrBeamPlugin #155: software update - freezes sometimes

clemniem, 2017-09-28:
Sometimes the software update freezes. But in the log it says successfull but the progress isnt saved, nor sent to frontend.

```
2017-09-28 09:13:14,723 - octoprint.plugins.softwareupdate - INFO - Starting update of mb-camera-calibration to 0cb790aaa17eea1e1caa5583a15af0a20abd72b6...
2017-09-28 09:14:11,839 - octoprint.plugins.softwareupdate - INFO - Update of mb-camera-calibration to 0cb790aaa17eea1e1caa5583a15af0a20abd72b6 successful!
2017-09-28 09:14:44,044 - octoprint.plugins.findmymrbeam - INFO - Registration to FindMyMrBeam. response: 200
```

clemniem, 2017-09-28:
I think I found the Problem:
When switching from feature branch back to develop i deleted also the 'current: XXX...XXX' entry in the config.yaml. After putting just some random numbers there it worked. So if the Commit is None, I got the error.

Gallore, 2017-10-04:
@clemniem what's your recommendation: fix or no fix? How does it relate to #163?

clemniem, 2017-10-04:
doesn't really relate to #163. Closed for now I guess let's see if it happens also during 'normal' updates.

closed 2017-10-04 by clemniem