mrbeam/MrBeamPlugin #157: workingArea, wrong size at first load

clemniem, 2017-10-02:
sometimes the working area isnt loaded correctly. Happened on my computer and on my friends. 
After reload everything is back to normal.
[image]

Gallore, 2017-10-04:
In this case only default machine profile is loaded but not MrBeamII profile.
Teja suggest to set the new MrBeamII as default.

clemniem, 2017-10-04:
Got it again and with this error message in console: 
``
Failed to load resource: the server responded with a status of 401 (UNAUTHORIZED) /plugin/mrbeam/profiles
``

Gallore, 2017-10-05:
Is this only when the user isn't logged it while the page is opened/loaded?

Gallore, 2017-10-05:
[image]

clemniem, 2017-10-05:
> Is this only when the user isn't logged it while the page is opened/loaded?

Hm.. maybe. But I think at least I was logged in when it happened. Or it stayed that way after log in at least.

Gallore, 2017-10-23:
Call '''http://mrbeam-f930.local/plugin/mrbeam/profiles''' was protected by @restricted_access. Therefor an anonymous user (read: user is about to log in) didn't get machine profiles and therefor the working area was set back to default size (read: Mr Beam I size)

commit d6c807de (2017-10-23): Fixes #157: Machine profiles accessible even for not logged in users
closed 2017-10-23 by Gallore