mrbeam/MrBeamPlugin #187: etc/mrbeam Device Info is None

clemniem, 2017-10-20:
There is no mrbeam device_info_file as expected in this function:
Should there be one?
```	
def _get_val_from_device_info(self, key):
  device_info_file = '/etc/mrbeam'
  try:
	with open(device_info_file, 'r') as f:
	  for line in f:
		line = line.strip()
		token = line.split('=')
		if len(token) >= 2 and token[0] == key:
			return token[1]
  except Exception as e:
		self._logger.error("Can't read device_info_file '%s' due to exception: %s", device_info_file, e)
  return None```

Gallore, 2017-10-20:
Yes should be. 
The file is created in newer images OR created by the Self-Check (USB-Stick)
I'll post an example in a second

Gallore, 2017-10-20:
pi@MrBeam-F930:~\> cat /etc/mrbeam
```
octopi=default Thu Sep 21 16:04:15 UTC 2017
hostname=MrBeam-F930
device_series=2C
device_type=MrBeam2C
serial=00000000E79B0313-2C
image_correction_markers=MrBeam2C-pink
```

clemniem, 2017-10-25:
There was the Issue once, where findmymrbeam sent answers with 200 , but the device did not show up on find.mr-beam.org nor through directly adressing.

```
2017-10-25 08:12:32,799 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-v1.3.4.3-py2.7.egg/octoprint/plugin/core.py:1399: UserWarning: No PKG-INFO found for package: octoprint_netconnectd
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2017-10-25 08:12:36,324 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-v1.3.4.3-py2.7.egg/octoprint/plugin/core.py:1399: UserWarning: No PKG-INFO found for package: octoprint_findmymrbeam
  warnings.warn('No PKG-INFO found for package: %s' % self.package_name)

2017-10-25 08:12:40,717 - octoprint.plugins.mrbeam - ERROR - Can't read device_info_file '/etc/mrbeam' due to exception: [Errno 2] No such file or directory: '/etc/mrbeam'
```

after creating an /etc/mrbeam manually it worked again.