mrbeam/MrBeamPlugin #193: Status locked, cannot home and no information to click

clemniem, 2017-10-30:
[image]

My laser is locked and I don't get the homing button. Also if I click on the questionmark next to the locked status, I don't get anything, nothing happens.

clemniem, 2017-10-30:
`helpers.js:1057 Error calling fromHistoryData on view model MotherViewModel : TypeError: self.state.isLocked is not a function
    at MotherViewModel.self._processStateData (http://mrbeam2-56d4.local/plugin/mrbeam/static/js/mother_viewmodel.js:299:24)
    at MotherViewModel.self._fromData (http://mrbeam2-56d4.local/plugin/mrbeam/static/js/mother_viewmodel.js:292:18)
    at MotherViewModel.self.fromHistoryData (http://mrbeam2-56d4.local/plugin/mrbeam/static/js/mother_viewmodel.js:288:18)
    at http://mrbeam2-56d4.local/static/js/app/helpers.js:1047:43
    at Pn (http://mrbeam2-56d4.local/static/js/lib/lodash.min.js:9:530)
    at Function.<anonymous> (http://mrbeam2-56d4.local/static/js/lib/lodash.min.js:30:66)
    at callViewModelsIf (http://mrbeam2-56d4.local/static/js/app/helpers.js:1041:7)
    at callViewModels (http://mrbeam2-56d4.local/static/js/app/helpers.js:1004:5)
    at Object.<anonymous> (http://mrbeam2-56d4.local/static/js/app/dataupdater.js:197:13)
    at fire (http://mrbeam2-56d4.local/static/js/lib/jquery/jquery.js:3187:31)`

Gallore, 2017-10-30:
Das ist sehr interessant, diese Fehlermeldung im MotherViewModel. Welche OctopPrint version hast du?

Gallore, 2017-10-30:
Talked to @clemniem : He had the nut yet supported version of Octoprint 1.3.5.1

closed 2017-10-30 by Gallore