mrbeam/MrBeamPlugin #208: Why is @firstrun_only_access disabled for rout '/calibration' etc?

Gallore, 2017-11-07:
There are three instances where the `@firstrun_only_access` is disabled. From my understanding it's security relevant that this is enabled, isn't it @hungerpirat ?

__init__.py: line 653:
```
	@octoprint.plugin.BlueprintPlugin.route("/calibration", methods=["GET"])
	#@firstrun_only_access
	def calibration_wrapper(self):
```

__init__.py: line 698:
```
	@octoprint.plugin.BlueprintPlugin.route("/take_undistorted_picture", methods=["GET"])
	#@firstrun_only_access
	def takeUndistortedPictureForInitialCalibration(self):
```

__init__.py: line 706:
```
	@octoprint.plugin.BlueprintPlugin.route("/send_calibration_markers", methods=["POST"])
	#@firstrun_only_access
	def sendInitialCalibrationMarkers(self):
```

clemniem, 2017-11-07:
Its disabled, so that without having someone logged in we can do some stuff from within OctoPrint, eg. the initial Camera Calibration. As far as I understood.

Gallore, 2017-11-07:
@clemniem : Nope. It's designed to allow exactly this kind of access to OP. Without the annotation it's alway open to everyone. I just talked to Teja and he confirmed this assumption.

Gallore, 2017-12-19:
`takeUndistortedPictureForInitialCalibration()` and `sendInitialCalibrationMarkers()` would need two decorators: `@firstrun_only_access` and `@restricted_access` but from my quick test just now, this combination doesn't work (at least not as expected).

amrsoll, 2020-07-02:
@Gallore - This seems to relate to what we want to reinstate, with the extra spice when inserting a support key that bypasses the restriction

closed 2021-03-11 by amrsoll