mrbeam/MrBeamPlugin #229: Configure Discovery Plugin for Windows

Gallore, 2017-12-05:
OctoPrints Discovery Plugin announces Mr Beam in local networks, so that it can be found by MrBeam-XXX.local. However, on Windows the device appears in 'Netzwerkumgebung' where more information are shown. 
Adjust these information from OctoPrint to MrBeam.

[image]

Apparently, "all" we need to do is to add proper values to config.yam. Sounds like a job for the image creation AND/OR plugins migration.

This is the code in OctoPrint that handles these values:
https://github.com/mrbeam/OctoPrint/blob/f1cf482919cd6745174763b8d9f14f5edfd8436c/src/octoprint/plugins/discovery/__init__.py#L118-L128

and here is the documentation:
http://docs.octoprint.org/en/master/bundledplugins/discovery.html

Gallore, 2017-12-13:
Get Discovery Plugin's output XML: mrbeam-f930.local/plugin/discovery/discovery.xml

viktorijad, 2017-12-13:
The name of the device is "OctoPrint instance" + the name of the device. This is due to the function in the link below:
https://github.com/mrbeam/OctoPrint/blob/f1cf482919cd6745174763b8d9f14f5edfd8436c/src/octoprint/plugins/discovery/__init__.py#L681

It should be changed to display only the name of the device. 

Furthermore, the number of the device should be changed automatically in config.yaml

viktorijad, 2017-12-18:
With writing 
```
		self._settings.global_set(["plugins","discovery","model","number"], "number hardcoded in plugin")
		self._settings.save()
```
here: https://github.com/mrbeam/MrBeamPlugin/blob/develop/octoprint_mrbeam/__init__.py#L105
we can change the model number of the device that is displayed.