mrbeam/MrBeamPlugin #271: Cannot update

r00tr4v3n, 2018-03-13:
> Updating...
Updating, please wait.
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Now updating mb_camera_calibration to 3500c77fa4cec319904d97b3ed834a4b2673d7e5
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python -m pip install git+ssh://git@bitbucket.org/mrbeam/mb_camera_calibration.git@3500c77fa4cec319904d97b3ed834a4b2673d7e5
Collecting git+ssh://git@bitbucket.org/mrbeam/mb_camera_calibration.git@3500c77fa4cec319904d97b3ed834a4b2673d7e5
Cloning ssh://git@bitbucket.org/mrbeam/mb_camera_calibration.git (to 3500c77fa4cec319904d97b3ed834a4b2673d7e5) to /tmp/pip-tBZoW8-build
Connection to bitbucket.org closed by remote host.
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.
Command "git clone -q ssh://git@bitbucket.org/mrbeam/mb_camera_calibration.git /tmp/pip-tBZoW8-build" failed with error code 128 in None
The update did not finish successfully. Please consult the log for details.

[octoprint.log](https://github.com/mrbeam/MrBeamPlugin/files/1808009/octoprint.log)

Gallore, 2018-03-20:
from the attached logfile:
```
2018-03-13 17:31:55,822 - octoprint.plugins.softwareupdate - ERROR - Update of mb-camera-calibration can not be performed
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-v1.3.6-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 908, in _perform_update
    update_result = updater.perform_update(target, populated_check, target_version, log_cb=self._log, online=online)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-v1.3.6-py2.7.egg/octoprint/plugins/softwareupdate/updaters/pip.py", line 81, in perform_update
    raise exceptions.UpdateError("Error while executing pip install", (stdout, stderr))
UpdateError
```

Gallore, 2018-03-20:
Is this error persistent? If yes, please reach out to support@mr-beam.org
Thank you.

Gallore, 2018-03-20:
Maybe related to #272 ?

Gallore, 2018-03-20:
Not reproducible here. And not the same as #272

r00tr4v3n, 2018-03-24:
It updated after several tries partially, showed errors instead of the software versions in the settings menu and finally finished an update completely.

closed 2018-03-24 by r00tr4v3n