mrbeam/MrBeamPlugin #302: Empty laserjob and exception in converter.py

Gallore, 2018-06-19:
This error happened with this svg file, sent in by a customer. He complained about something different. However, this error just happened to me.

I put the red line on skip and the blue line on engrave.
[Bürgersteigfliesen2.svg.zip](https://github.com/mrbeam/MrBeamPlugin/files/2115537/Burgersteigfliesen2.svg.zip)

```
018-06-19 14:15:34,854 - octoprint.plugins.mrbeam - ERROR - Conversion error (I/O operation failed.):
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/__init__.py", line 1330, in do_slice
    engine.convert(on_progress, on_progress_args, on_progress_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/gcodegenerator/converter.py", line 288, in convert
    self.export_gcode()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/gcodegenerator/converter.py", line 691, in export_gcode
    shutil.move(self._tempfile, destination)
  File "/usr/lib/python2.7/shutil.py", line 302, in move
    copy2(src, real_dst)
  File "/usr/lib/python2.7/shutil.py", line 130, in copy2
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 82, in copyfile
    with open(src, 'rb') as fsrc:
IOError: [Errno 2] No such file or directory: '/tmp/_converter_output.tmp'
```

hungerpirat, 2018-08-28:
Tried to reproduce without success. Conversion of the svg works without errors, result looks fine. 
According to the error message I assume, there was a sd-card write failure. 
Is there any information available on which machine you tested this @Gallore ?

Gallore, 2019-08-30:
Not reproducible ! Close it

closed 2019-08-30 by Gallore