mrbeam/MrBeamPlugin #304: GCODE handling of G0 commands

Gallore, 2018-06-28:
Looks like G0 command are handled buggy by grbl.

Example 1:
Place a SVG file (e.g. MrBeam logo; not a QuickShape!) on working-area, start a laserjob. While the laserhead moves to it's first position, press pause, then unpause. Laser won't turn on when it should.
https://mr-beam.zendesk.com/agent/tickets/299

Example 2:
Use the GCODE below. It's a manually modified GCODE which took two QuickShapes as base. 
During both G0 commands the laser is on. During the first G0 I'm not even able to pause the system. 

```
;Generated from temp.svg 3b5510f21c83e60b3fab46a3d4c5e8ec08d93114
; gc_nexgen gc_options: beamOS:0.1.40 on MrBeam-ANDY, gc_nextgen:0.1, enabled:true, precision:0.05, optimize_travel:true, small_paths_first:true, clip_working_area:false, clipRect:0,0,500,390
; created:2018-06-28 17:55:09
; laser params: {u'#e25303': {u'passes': u'1', u'feedrate': u'1375', u'color': u'#e25303', u'job': 0, u'intensity_user': u'100', u'intensity': 1300, u'pierce_time': u'0'}}
; Layer:?, outline of:pathSjiyua1rr7d, stroke:#e25303, {u'passes': u'1', u'feedrate': u'1375', u'color': u'#e25303', u'job': 0, u'intensity_user': u'100', u'intensity': 1300, u'pierce_time': u'0'}
; pass:1/1
;_gc_nextgen_svg_id:wa_cacila-0,node:path,mb:color:#e25303,mb:id:wa_cacila-0,clip_working_area_clipped:false

F1375;#e25303
M03 S1300;#e25303

G0X359.78Y314.10

G1X459.78Y314.10
G1X459.78Y264.10
G1X359.78Y264.10
G1X359.78Y314.10
M05

; Layer:?, outline of:pathSjiyua1rr7d, stroke:#e25303, {u'passes': u'1', u'feedrate': u'1375', u'color': u'#e25303', u'job': 0, u'intensity_user': u'100', u'intensity': 1300, u'pierce_time': u'0'}
; pass:1/1
;_gc_nextgen_svg_id:wa_walowo-0,node:path,mb:color:#e25303,mb:id:wa_walowo-0,clip_working_area_clipped:false

F1375;#e25303
M03 S1300;#e25303

G0X79.89Y85.61

G1X179.89Y85.61
G1X179.89Y35.61
G1X79.89Y35.61
G1X79.89Y85.61
M05

M05

```

hungerpirat, 2018-09-05:
In example 2, the gcode is clearly wrong. Good thing, I can't reproduce that anymore (Mr Beam Plugin: v0.1.17, OctoPrint: v1.3.6.2, Image fingerprint: None, Software tier: DEV, Environment: DEV)
Trying to reproduce example 1: no success (which is a good thing).

@Gallore are you able to reproduce that on some machine ?

Gallore, 2019-08-30:
Never happened anymore!!!!!

closed 2019-08-30 by Gallore