mrbeam/MrBeamPlugin #401: Real time image adjustments sliders in the working area

Gallore, 2019-02-18:
teja implemented the prototype on the weekend. Now bring this into development by following the TODOs further down.

* branch: f_realtime_img_adjust
* TODOs:
    * remove switches from conversion dialog
    * limit to 90% and/or maybe change contrast slider to increase-option only
    * re-enable quick text font copying
        * test if copying the font defs into the svg even though no quick text is used brings a relevant speed up
    * end-to-end test
    * add sharpening slider
    * try gamma-correction slider
    * clean up backend (remove unneeded code)
    * end-to-end test
* Links:
    * https://www.html5rocks.com/en/tutorials/canvas/imagefilters/

irlaec, 2019-03-25:
[image] [(3) Image adjustment integrated in develop](https://trello.com/c/IVrj1Ylr/36-3-image-adjustment-integrated-in-develop)

irlaec, 2019-04-02:
#442

closed 2019-04-02 by irlaec