mrbeam/MrBeamPlugin #444: Make GRBL flashing per serial more stable

Gallore, 2019-04-02:
Two things should be done to improve stability of flashing grbl over serial bus as we do from OctoPrint:
- stop iobeam or make sure it doesn't send/read on i2c bus during the auto-reset phase of avr-dude
- increase wait time from 100ms to 300ms in /usr/bin/autoreset (reset function)

Here are Bastis comments on the subject:

> Ich hab den Reset gefunden und die dazugehörigen Timingprobleme auch also das andere die auch haben.
> https://github.com/deanmao/avrdude-rpi

```
def reset():
    time.sleep(0.05)
    i2c.write_byte_data(i2c_addr, rst & 0xff, (rst >> 8) & 0xff)
    time.sleep(0.10)
    i2c.write_byte_data(i2c_addr, nom & 0xff, (nom >> 8) & 0xff)
```

> irgend eine uraltversion davon benutzen wir, das ist der Modified AVRDUDE vom Flo
> Das ist der Aufruf in usr/bin/autoreset der über den I2C Bus den Resetpin auslöst, die Resettime ist da auf 0.1s , im Git ist es 0,13 und die mit den Problemen sind auf 0,3 hoch um sie loszuwerden.
> https://github.com/deanmao/avrdude-rpi/issues/4
> auf dem Beam ist der Timingfile   /usr/bin/autoreset  das Delay kann von 0,1 auf 0,3 s erhöht werden und die initiale Pause kann gestrichen werden.

According to Basti the afore mentioned function should look like this:
```
def reset():
    # remove initial wait time, it's worth nothing
    i2c.write_byte_data(i2c_addr, rst & 0xff, (rst >> 8) & 0xff)
    time.sleep(0.30) # 300ms instead of 100ms
    i2c.write_byte_data(i2c_addr, nom & 0xff, (nom >> 8) & 0xff)
```

Sebastian-MrBeam, 2019-04-09:
I changed to several resets with short delays , because of a hardware bug.
The reset line is only triegerd in the moment of setting a value not by holding it stable.
So the reset command has to be set several times over the 0.3s periode.
 

[autoreset.txt](https://github.com/mrbeam/MrBeamPlugin/files/3058869/autoreset.txt)

Gallore, 2019-05-02:
[image] [GRBL programmer script](https://trello.com/c/khs1WPgQ/102-grbl-programmer-script)

closed 2019-05-15 by irlaec