mrbeam/MrBeamPlugin #458: Job just stops, no reaction from OneButton

Gallore, 2019-04-17:
[octoprint.log](https://github.com/mrbeam/MrBeamPlugin/files/3088910/octoprint.log)

Last 'valid' line from job:
`2019-04-17 09:21:36,097 - octoprint.plugins.mrbeam - INFO - on_event() PrintProgress: {'progress': 100, 'time': 1099.3358399868011}`
Everything seems fine however nothing more happened. Once OneButton got pressed, device went into an endless loop which had no visible reaction to the user.

Gallore, 2019-05-02:
[image] [Bug fix](https://trello.com/c/bNMAieg8/108-bug-fix)

irlaec, 2019-07-02:
Probably the same error: the job was done but the Mr Beam did not recognize it. The laserhead did not go back to the homing position and the timer of the job was still running, as well as the fan. 
[octoprint.log](https://github.com/mrbeam/MrBeamPlugin/files/3349901/octoprint.log)

closed 2020-02-19 by Gallore