mrbeam/MrBeamPlugin #601: iobeam: correct bug in FanPCB

Gallore, 2019-10-10:
If dust value passes 585, the fan will be turned down to min_pwn if fanPCB is in auto mode due to an buffer overflow.

Septs sto du:
If client (= OctoPrint) sets auto mode AND dust value is higher than 500, set fanPCB to fixed value of 100; once dust value goes below 450, switch back to auto mode. 

For the client this should be invisible: fan_dynamic should still report state "auto" because from the clients view this is still the case because if the dust value drops, the fan will be turned down eventually.

Gallore, 2019-10-10:
[image] [iobeam bug fix for fanPCB](https://trello.com/c/Kq2vIj7B/283-iobeam-bug-fix-for-fanpcb)

amrsoll, 2019-11-29:
Fixed in https://bitbucket.org/mrbeam/iobeam/pull-requests/42

closed 2019-11-29 by amrsoll