mrbeam/MrBeamPlugin #712: text cropped during rendering

hungerpirat, 2020-01-18:
How (hard) to reproduce:
1. place quick text with 10+ characters and non-default font, e.g. "acdefghijk" with font "comfortaa". 
2. scale it up, e.g 400 mm width.
3. convert it - place gcode afterwards on the working area and see the cropped edges.

- happens more often in fresh started browsers
- happens reproducable only one time - second time works fine.

[image]

hungerpirat, 2020-01-18:
Investigation log: 
https://github.com/mrbeam/MrBeamPlugin/blob/develop/octoprint_mrbeam/static/js/render_fills.js#L12- This call returns *sometimes* a bad (means too small) bounding box.
- Wondering if there is some JS optimization in the background, which leads to the correct behaviour on the second try.
- after adding debug statements, I wasn't able to reproduce it at all. Each of those debug outs called getBBox() on the svgElement during the rendering preparation. Removing the debug statements didn't lead to the former behaviour. Restarting the browser did somehow.

hungerpirat, 2020-01-18:
[image]

Here's probably the cause: 
- Snap initiates during the getBBox() call a http call to fetch the dataurl encoded font. 
- somehow this call is delayed sometimes. 
- in this cases the getBBox() can't properly calculate the bbox size due to the missing font.

Gallore, 2020-04-15:
https://mr-beam.freshdesk.com/a/tickets/18357

Customer still complained about this bug.
G-Code attatched.
> Rechteck_2more.gco: Fehlerhafte Datei, Font: Allerta Stencil
> Rechteck_2more.1.gco: Fehlerfreie Datei, Font: Roboto

Chome: 80.0.3987.163 (Offizieller Build) (64-Bit) unter macOS 10.14.6
Mr Beam Plugin: 0.6.2 (current Prod)

[Archive 2.zip](https://github.com/mrbeam/MrBeamPlugin/files/4483368/Archive.2.zip)

commit c6742976 (2020-01-18): fixes #712