mrbeam/MrBeamPlugin #728: v0.5.5.1 BETA: Image Separator lost components of the image

Gallore, 2020-02-05:
v0.5.5.1 BETA:

Letters "o" and "e" got lost.
Everything should be engraved only.

G-Code preview:
[image]

How the G-Code actually looks like:

[image]
(on https://nraynaud.github.io/webgcode/ )

[SVG&GCO.zip](https://github.com/mrbeam/MrBeamPlugin/files/4161175/SVG.GCO.zip)

hungerpirat, 2020-02-08:
[image]
Not reproducable - I tried __many__ different ways, even I installed his used font "Bahnschrift".

hungerpirat, 2020-02-08:
Colateral findings: 
Not converted fonts work fine if they are installed on the local computer. 
As the browser does not know which fonts are installed - we just might ask the user if he gives us a list?