mrbeam/MrBeamPlugin #749: SVG split does not work properly

irlaec, 2020-02-25:
If the svg contains too many different "pieces" and you try to split it "into shapes", it just crashes. Try with the following file: 
[konfetti.zip](https://github.com/mrbeam/MrBeamPlugin/files/4251120/konfetti.zip)
I also think that it wouldn't make sense to split a svg in more than ~10 pieces because otherwise the designs list gets too big.

I also had problems with the vertical and horizontal splitting, but I can't reproduce them. With the same svg, it happened that the splitting didn't seem to be logical (like one of the pieces was way bigger than the other one for example).

In general I think that it just needs some more testing and polishing. The idea is to just hide it from the users until it's finalized and then make it visible again and present it in the next what's new screen.

irlaec, 2020-06-02:
[image] [Fix: SVG Split - finalize (Teja?)](https://trello.com/c/e4kdHIRx/520-fix-svg-split-finalize-teja)

hungerpirat, 2020-06-09:
fixed. @irlaec please review.

closed 2020-06-30 by irlaec