mrbeam/MrBeamPlugin #780: Overshoot creates visible marks on some materials

Gallore, 2020-03-13:
On some materials (leather) overshoot creates visible marks. (See image - gco attached as zip)
Current suggested workaround: set an pierce time of 1ms, this will disable overshoot.

However, we should come up with a full solution. This could wither be:
- add a switch to the advanced settings and add it to the material settings accordingly
- find a metric of how to turn it on/off automatically

[image]
[basst_v1.1.gco.zip](https://github.com/mrbeam/MrBeamPlugin/files/4329793/basst_v1.1.gco.zip)

Gallore, 2020-04-07:
[image] [Bug: Overshoot creates visible marks on some materials](https://trello.com/c/sUELblya/498-bug-overshoot-creates-visible-marks-on-some-materials)

amrsoll, 2020-09-02:
Laserhead 1st has to stop before turning off -> turning off the laserhead won't solve the problem.
2nd options: keep momentum by doing circles at the end of each line.

amrsoll, 2020-10-24:
GRBL 1.1 Update summary

>- **Laser Mode** : The new "laser" mode will cause Grbl to move continuously through consecutive G1, G2, and G3 commands with spindle speed changes. When "laser" mode is disabled, Grbl will instead come to a stop to ensure a spindle comes up to speed properly. Spindle speed overrides also work with laser mode so you can tweak the laser power, if you need to during the job. Switch between "laser" mode and "normal" mode via a `$` setting.
>	- **Dynamic Laser Power Scaling with Speed** : If your machine has low accelerations, Grbl will automagically scale the laser power based on how fast Grbl is traveling, so you won't have burnt corners when your CNC has to make a turn! Enabled by the `M4` spindle CCW command when laser mode is enabled!

Making the overshoot feature obsolete.