mrbeam/MrBeamPlugin #848: SVG circle (and other) without ID: do better handling

Gallore, 2020-05-28:
A plain circle within a SVG was not lasered / empty gcode was generated.
This circle node had no ID on the working area. (Don't know why.)
[circle_test.svg.zip](https://github.com/mrbeam/MrBeamPlugin/files/4695877/circle_test.svg.zip)

Looks like this code change caused embed_gc() method to skip such a node:
https://github.com/mrbeam/MrBeamPlugin/commit/12c20d035a72ad5f6e992be097473e8405b835a4
I documented this code change very badly (not at all). I just remember that there was an element within a customer's SVG that caused a crash in embed_gc(). (Apparently becasue of no id)

Thoughts:
- Why/how does it happen that there are elements without ID on the working area. I think this should be avoided.
- What to do in embed_gc in case thatere is no ID. (`mb_meta[id]` seem to require this id)
- How to tell/show the suer if this new (#847) try?catch inn embed_gc() catches an exception? Just pass on? PNotify? Maybe it would be best to remove the according element from the working area so that the users sees what's missing. For that we need  to run embed_gc() already when we place an SVG on the working area (it takes very little time even on big files.) in order to make the removal visible to the user.

Gallore, 2020-05-29:
PR #851 fixes (again) the original problem that happens if there is an element w/o id.

Reproduce with this SVG: 
[2020-perlen-laser-blau-12mm.svg.zip](https://github.com/mrbeam/MrBeamPlugin/files/4702843/2020-perlen-laser-blau-12mm.svg.zip)

amrsoll, 2020-07-02:
So this issue can be closed?