mrbeam/MrBeamPlugin #862: Add a warning if user uploads a file with existing name.

Gallore, 2020-06-12:
At the moment the existing file os overwritten without any question. Then we might face caching issues where still the old file is put on the working area,

Jessie:
> Hab jetzt was gehabt beim Mr Beam und zwar hab ich eine Datei “Schlüsselanhänger” genannt und sie hochgeladen. Jetzt hat er die Original-Datei angezeigt, die bei jedem Mr Beam sowieso schon drauf ist. Ist jetzt halb so wild weil es ein banaler Name ist. Kann mir aber vorstellen, dass man schon öfters was “test” oder “final” nennt. Wär cool wenn mich die Software darauf hinweist, dass eine Datei in meinen Dateien schon so heißt :leichtes_lächeln: Hab mich jetzt gewundert beim 3. mal warum es nicht richtig hochlädt :lächeln:

amrsoll, 2020-07-02:
See #141

irlaec, 2020-08-18:
[image] [Add a warning if user uploads a file with existing name. #862](https://trello.com/c/DE4MQD2u/604-add-a-warning-if-user-uploads-a-file-with-existing-name-862)