mrbeam/MrBeamPlugin #873: Laser head stops moving when opening a big library

amrsoll, 2020-07-02:
**Problem**

When opening a very populated library, it has been observed that the laser head would jitter instead of moving smoothly during a laserjob because of bottle-necked gcode streaming.

**Recommandation(s) for resolution**

- Decrease the niceness of the GRBL streamer, or increase the niceness of all other tasks during a laserjob.
- Limit the number of files displayed simultaneously in the library.

An additional message could be shown in the library during a laserjob
>A laserjob is currently running. The design library will run slower than usual.

Gallore, 2020-09-01:
[image] [Laser head stops moving when opening a big library](https://trello.com/c/lvjDFppA/633-laser-head-stops-moving-when-opening-a-big-library)

amrsoll, 2020-09-22:
Currently tasked with investigation slow Front End loading screen 

#### Device

OP v1.3.6  +  [MrBeamPlugin], netconnectd

Running on a RaspberryPi 3 with Ethernet connection

#### What we have experienced

The stutter has been mostly observed when loading up a crowded design library (via storage manager) when over the 150 files mark (I'm testing with ~200 files)   

Attached is the log and pstats files when running the code with a profiler after cleaning the metadata, loading the interface and letting the storage manager run in the background.

Loading the interface from scratch consistently takes ten's of seconds more than usual, and the OP doesn't seem to recognise the connection status of the user very well. 

#### Suspects

Given the age of this issue, it ain't easy - but it points me in the direction of limited I/O and the storage manager.

After some analysis, I have found that the storage manager with a large library is quite expensive between the use of yaml.dump and yaml.load (no news here), as well as dict_merge - used to update `.metadata.yaml` each time a file is processed. Granted, it's not quite as problematic when the `.metadata.yaml` has been built. 

#### Possible mitigations

Making the [GCode analysis tool sleep longer](https://docs.octoprint.org/en/1.3.12/configuration/config_yaml.html?highlight=profiler#gcode-analysis) should help ~ Doesn't help with the initial queue build-up (when starting OP).

I have written a small OP patch (attached) that allows dict_merge to run in place because of it's aforementioned presence on the profiler. - I've also written it for the latest OP/develop if interested. The in-place dict_merge calls were on average 2~3 times faster.

[octoprint_load_front_long.log](https://github.com/mrbeam/MrBeamPlugin/files/5264291/octoprint_load_front_long.log)
[octoprint_load_front_long.pstats.zip](https://github.com/mrbeam/MrBeamPlugin/files/5264302/octoprint_load_front_long.pstats.zip)
[in_place_merge.diff.txt](https://github.com/mrbeam/MrBeamPlugin/files/5264307/in_place_merge.diff.txt)

Perhaps that could be of interest relating to OctoPrint/Octoprint#1241 @foosel ?

foosel, 2020-09-28:
OctoPrint hasn't been using YAML for metadata storage in a long while, due to observed performance issues with that. It's all json now. Really good idea with the in place merge though, if that's ok with you I'll add that on my end as well.

foosel, 2020-09-28:
Have factored that into OctoPrint's `maintenance` branch and will make it part of 1.5.0. Small note: Your patch up there doesn't set the parameter on the recursive call, so subsequent recursions will still lead to deepcopy calls.