mrbeam/MrBeamPlugin #948: Camera marker warning: does not show right after opening the lid

Gallore, 2020-08-18:
Reproduce:
- Cover some markers and wait til the pink circle show on the working  area. (as expected)
- Close the  lid: the markers disappear as expected. However, the warning in the preview pane remains (not sure if this is  a bug or a feature.)
- Open the lid again. The markers shoudl come back and flash again, but they do NOT. Only when you change the content of the image, an new image is processed which will trigger the markers to flash again.

Todo:
Decicide how it should be?
- Should the markers keep flashing while the lid is closed but the shown image might suffer from inaccuracy?
- Should the waring in the preview pane remain while the lid is  closed?
- when exactly should markers and warnings re-appear?

Gallore, 2020-08-18:
[image] [Camera marker warning: does not show right after opening the lid](https://trello.com/c/ppXnpcIi/602-camera-marker-warning-does-not-show-right-after-opening-the-lid)

amrsoll, 2020-08-18:
I would go an other route:
- Show a loading spinner while MrBeam is still processing the image
- Continue showing the warning as long as the picture shown wasn't taken with all the markers detected (nothing to change)
- Pink markers could be slightly different when reopening the MrBeam - Let the user know that there was a problem for the current image, but it might ok for the coming picture.

amrsoll, 2020-08-31:
Decision:
- Show gray transparent marker when lid is closed up to the next picture received from the backend (no flashing).

Gallore, 2020-09-01:
[image] [Camera marker warning: does not show right after opening the lid](https://trello.com/c/LyHO5sdo/632-camera-marker-warning-does-not-show-right-after-opening-the-lid)

PR #975 (2020-09-08): Show missing markers in gray when camera is inactive.
Fixes #948

Previously missed markers are not stored in backend -> Reboot or even refresh will get rid of them and the warning.

I kept it quite discrete, no blinking.
[image]
closed 2020-09-08 by amrsoll