mrbeam/MrBeamPlugin #949: Bug: pierce time and progressive option do not show consistently if 2 or more cutting jobs

Gallore, 2020-08-18:
@khaledsherkawi Please add more details

Gallore, 2020-08-18:
[image] [Bug: pierce time and progressive option do not show consistently if 2 or more cutting jobs](https://trello.com/c/NsNAWhDM/603-bug-pierce-time-and-progressive-option-do-not-show-consistently-if-2-or-more-cutting-jobs)

khaledsherkawi, 2020-08-26:
- Channel: Stable
- Affected version: 0.7.1 (and below)
- Component: Front-end
- Browser: Chrome
- Priority: Minor
- Description: While dealing with more than one cutting color in lasering and whenever you want to assign a different set of parameters to each color, some issues surface regarding the "advanced settings" parameters. These "advanced settings" parameters will be showing for one or more cutting colors and not for others in certain scenarios described below.
- Steps to reproduce:
1. GIVEN you added 3 shapes with 3 different line colors

Gallore, 2020-09-01:
[image] [Bug: pierce time and progressive option do not show consistently if 2 or more cutting jobs](https://trello.com/c/XC4SfC7l/631-bug-pierce-time-and-progressive-option-do-not-show-consistently-if-2-or-more-cutting-jobs)