mrbeam/MrBeamPlugin #1012: Exception mrb_check while trying to upload without internet connection

Gallore, 2020-09-26:
Please investigate

```
2018-01-12 19:24:08,337 - mrb_check.tools.upload - INFO - Uploading Data to Google... (try: 0)
2018-01-12 19:24:08,342 - urllib3.connectionpool - INFO - Starting new HTTPS connection (1): script.google.com
2018-01-12 19:24:08,345 - mrb_check.tools.upload - ERROR - Upload (try:0) ERR: ConnectionError. No internet connection?
2018-01-12 19:24:08,346 - mrb_check.tools.upload - INFO - Uploading Data to Google... (try: 1)
2018-01-12 19:24:08,350 - urllib3.connectionpool - INFO - Starting new HTTPS connection (1): script.google.com
2018-01-12 19:24:08,353 - mrb_check.tools.upload - ERROR - Upload (try:1) ERR: ConnectionError. No internet connection?
2018-01-12 19:24:08,353 - mrb_check.tools.upload - INFO - Uploading Data to Google... (try: 2)
2018-01-12 19:24:08,358 - urllib3.connectionpool - INFO - Starting new HTTPS connection (1): script.google.com
2018-01-12 19:24:08,361 - mrb_check.tools.upload - ERROR - Upload (try:2) ERR: ConnectionError. No internet connection?
2018-01-12 19:24:08,362 - mrb_check.tools.upload - ERROR - Uploading Data to Google: ERR: -1 None
2018-01-12 19:24:08,367 - mrb_check.tools.upload - INFO - save_data_to_file() save upload data to file: /tmp/mrb_check/mrb_check_failed_upload_MrBeam-2B98_00000000800BC034-2U_1515785048.json
2018-01-12 19:24:08,371 - mrb_check.__init__ - ERROR - Exception during test execution:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/mrb_check/__init__.py", line 238, in run
    failed_uploads_dir=self.failed_uploads_dir)
  File "/usr/local/lib/python2.7/dist-packages/mrb_check/tools/upload.py", line 59, in upload
    save_data_to_file(data, failed_uploads_dir=failed_uploads_dir)
  File "/usr/local/lib/python2.7/dist-packages/mrb_check/tools/upload.py", line 145, in save_data_to_file
    logging.getLogger(__name__).exception("save_data_to_file() Exception in while writing upload data to external folder: '%s'", log_dir)
NameError: global name 'log_dir' is not defined
```

Gallore, 2020-09-29:
[image] [Exception mrb_check while trying to upload without internet connection](https://trello.com/c/CADU7XHH/656-exception-mrbcheck-while-trying-to-upload-without-internet-connection)

amrsoll, 2020-09-30:
Issue was already fixed 25 days ago - please update your mrb_check

closed 2020-09-30 by amrsoll