mrbeam/MrBeamPlugin #1013: Cornercalibration: SHow warning if user tries to start engravin without homing  cycle

Gallore, 2020-09-26:
If the device is not homed, it's possible to click the "Engrave calibration markers" button in the user corner calibration page. Of course it fails but it does not tell the user why or how to solve the issue. There should be a warning telling the user that he needs to to a homing cycle first.

hungerpirat, 2020-09-26:
I'd suggest to just do the homing cycle automatically instead requesting
additional steps from the user.

Gallore, 2020-09-29:
[image] [Corner calibration: Show warning if user tries to start engraving without homing cycle](https://trello.com/c/gk00BDkO/655-corner-calibration-show-warning-if-user-tries-to-start-engraving-without-homing-cycle)

PR #1019 (2020-10-01): warning message: perform homing before arrow engraving.
Fixes #1013 

[image]
closed 2020-10-01 by amrsoll