mrbeam/MrBeamPlugin #1033: Write settings the way they are supposed to be.

amrsoll, 2020-10-08:
PR #1031 introduces a (much) better settings syncing with the front-end - it should become even better with OP 1.4

It would be best to change all of our settings management to do the same thing and - in the end - no longer use on_settings_save to save the value of a setting.

See 
- Commit https://github.com/mrbeam/MrBeamPlugin/pull/1031/commits/d5caf5bd91c96c0e4fbb2d632f39b2ff02de2f71 
- [OctoPrint-DisplayLayerProgress](OctoPrint-DisplayLayerProgress)
  [js](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/tree/abceeb49d6c598e07ff7143f78c422f6ebc335cc/octoprint_DisplayLayerProgress/static/js)
  [on_settings_save](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/blob/abceeb49d6c598e07ff7143f78c422f6ebc335cc/octoprint_DisplayLayerProgress/__init__.py#L1407)

On the example, on_settings_save is supposed to act on the values that will be saved.

amrsoll, 2020-10-08:
@Gallore I believe you are the best suited to make sure the different settings get rewritten correctly?

irlaec, 2020-10-15:
[image] [Redo find.mr-beam frontend](https://trello.com/c/NH7be2UH/681-redo-findmr-beam-frontend)