mrbeam/MrBeamPlugin #1042: Camera Settings: Page show "not all markers found" and no camera images after page reload while lid open

Gallore, 2020-10-14:
Only tested on v0.7.8:
Reproduce:
Open the lid, wait for the cam to show the latest image, reload frontend, go to camera settings page:
Staus warns: "Nicht alle pinken Markierungskreise wurden erkannt. Fehlende Markierungen: Oben links, Unten links, Unten rechts, Oben rechts" and there are no camera images (prepared and raw image) shown, eventhough the camera is working correctly. As soon as you change something on the working area, the status message is updated and new images are shown in the settings page.

amrsoll, 2020-10-14:
So it's a semi-expected behaviour, which is a bug now that we don't stop and restart the camera with client (dis)connections.

TODO : `onStartupComplete()` request the camera status and the latest found markers.

Do you think it would be a good idea to wrap this together with #1043 ?
>Create separate Status ViewModels

Gallore, 2020-10-20:
I think it should be handled separately becasue of the different prios. This on is a bug that should be fixed soon (and seems an easy fix to me), while the status object is a medium refactoring that needs more testing and has in general lower prio.