mrbeam/MrBeamPlugin #1050: Insure the correct Laser cutter Profile is used.

amrsoll, 2020-10-27:
What is the best option to make sure we have the correct profile in a new device? This includes new images that are used for repairs on older devices.

## Status Quo

The laser profiles can be found as yaml files in a folder under `.octoprint/printerprofiles/lasercutterprofiles/`. The plugin uses the name of the device found in `/etc/mrbeam` to chose the correct profile. If the device is a `MrBeam2U` model, and the `MrBeam2U.profile` is missing, then it only uses the `my_default.profile` - which can have the wrong settings. 

## Resolutions
### Hard coded into the plugin (or other repo)

Not a bad strategy. It might be best to keep the configs in a separate repo for our sanity.

Advantages

- Easy to maintain / change / update
- Any overwriting settings file can partially overwrite the hard-coded defaults

Disadvantages

- CodeBase increase

### Based on the environment

Using the different available 

Advantages

- In theory - No need to tell the device which model it is. 

Disadvantages

- Not all model have a unique environment.
- Increased Prototyping complexity with behaviour depending on what is available for the machine.

### Add profiles to the update process

This is equivalent to adding a new repo containing the profiles. The setup script would put them into the correct folder on the device.

## Bonus objective

### GUI device configuration

In **Support mode**, one could :

- Change the device model and name
- Add changes on top of the default settings

amrsoll, 2020-10-27:
[image] [Failsafes for Lasercutter profiles.](https://trello.com/c/I37sAL4u/619-failsafes-for-lasercutter-profiles)

Gallore, 2020-11-10:
I prefer the first option because it's the simplest option: Hard coded into the plugin
This way it's easiest to maintain (update) and make sure that every device has the correct profile available. it's no longer possible, that the profile is not installed. All other options (including separate repo) adds more complexity to something that gives us very little value.
With Mr Beam III we might want to refactor this.

Gallore, 2020-11-10:
Once we switched we should consider deleting the existing profiles and als remove them from the self_check sticks and other usb sticks.

closed 2021-01-26 by amrsoll