mrbeam/MrBeamPlugin #1088: Save file with unicode characters

amrsoll, 2020-11-26:
It seems like there can be unicode characters outside the text elements as well. Each time we save unsanitised file contents, we should do it with a binary write method.

```
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/filemanager/__init__.py", line 31, in save
     d.write(self.content)
 UnicodeEncodeError: 'ascii' codec can't encode character u'\xfc' in position 339207: ordinal not in range(128)
```

Taken from this stacktrace:

```
14:15:40 octoprint ERROR Exception on /plugin/mrbeam/convert [POST]
 Traceback (most recent call last):
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
     response = self.full_dispatch_request()
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
     rv = self.handle_user_exception(e)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
     reraise(exc_type, exc_value, tb)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
     rv = self.dispatch_request()
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
     return self.view_functions[rule.endpoint](**req.view_args)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1114, in decorated_view
     return flask.ext.login.login_required(func)(*args, **kwargs)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.11-py2.7.egg/flask_login.py", line 758, in decorated_view
     return func(*args, **kwargs)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/__init__.py", line 1755, in gcodeConvertCommand
     file_name=history_filename, content=content
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/filemanager/__init__.py", line 63, in add_file_to_design_library
     allow_overwrite=True,
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/__init__.py", line 475, in add_file
     path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=allow_overwrite, display=display)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/filemanager/storage.py", line 762, in add_file
     file_object.save(file_path)
   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_mrbeam/filemanager/__init__.py", line 31, in save
     d.write(self.content)
 UnicodeEncodeError: 'ascii' codec can't encode character u'\xfc' in position 339207: ordinal not in range(128)
```

hungerpirat, 2020-11-27:
isn't this the bug @Gallore fixed few weeks ago ?

hungerpirat, 2020-11-27:
talking about https://github.com/mrbeam/MrBeamPlugin/commit/a57a6ffd185df920cfddeb20f2a9b2ac975230d6

amrsoll, 2020-11-28:
No it's not. This is about some of the content of the file and the `FileManager.File.write()` expects ascii input.

PR #1089 (2020-11-30): Fix unicode file write with unicode content.
Fixes #1088

Also a `with open(path) as f` implies closing the file on exit of the `with` statement.
commit 07c225df (2020-11-30): Fix unicode file write. (#1089)

Fixes #1088
commit 18f11df8 (2020-12-01): v0.7.12 BETA release (#1093)

* v0.7.11.1: version Bump for develop

* Fixes bug where .MRB files did not get sanitized. (critical)

* FIx: Extra-Overshoot Button should have "Fast" as preselected

* Missing Pause Screen: potential fix in mutationCallback()

* Logging of frontend ajaxError: Avoid endless loops

* Fix cli tool to undistort the raw pictures.

* Maintenance values adjusted (#1085)

* Logging issues during conversion (try/catch) (#1090)

* Fixes logging issue where we did not see an exception in analytics

* Added more try/catch to /convert call to get exceptions in analytics

* F [... cut]
closed 2020-11-30 by Gallore