mrbeam/MrBeamPlugin #1137: Add more data to user review (NPS)

Gallore, 2021-02-02:
When we submit a user review ("Would you recommend Mr Beam to your friends? 1-10?") we need to add more device data and also ask the user more details.

Device data:
- SW version (mr beam plugin)
- model (MRBEAM2_DC, ...)
- production date (be aware that older devices do not have a model_id or production date)

Server side:
We need to add the date when the review was uploaded (which might be very different from the date when the review was done.) The creation date of the json file in the GCP bucket could be used. All following components need to user or forward this date.

User data:
We should ask the user if it's ok to contact her/him "in case we have questions" and offer to add email address or phone number. Could be one (free text) field or two fields. We could also pre-fill the user's email address, but I lean towards not doing this. We probably also need a statement what we use these data for.

Gallore, 2021-02-03:
Device data and server side fixed in commit 915d4d8

Gallore, 2021-02-05:
@khaledsherkawi please focus un the part "User data" in the description. The other things are already done.

Gallore, 2021-02-08:
@khaledsherkawi Please KEEP the switch for users who rated 7 and better!
We need a switch AFTER the suer entered his feedback and submitted it:
- <7 "Thank you for your feedback"
- >= 7: "Thank you for your feedback" + Please give us a review. For now this screen could be simple - just what we have now. Bonus points if the call to action (the review link) is a bit more visible. Later on (probably not for this release) we redesignt this scree according to how I described it in the description.

Gallore, 2021-02-10:
Wording:

 **Screen 1**

Title: 
`How likely are you to recommend Mr Beam to a friend or colleague?`
Please add the text "Not at all likely" to the 1 button and "Extremely likely" to the 10 button. Ideally on top or left/right. Should not be too big.

**Screen 2**

Title:
`Thank you for your response.`

Label Feedback field: 
`Help us by explaining your score...`

Label contact field: 
`Phone or email: (optional)`
I think the word "optional" should be as bold as the "Phone or email" is

Contact information field explanation:
`If it's ok to contact you about your feedback.`
muted - could be within the same line as the label - with this one I'm not 100% sure.

Button: 
`Send`

I took my ideas from there:
https://www.wootric.com/blog/what-is-nps-the-concept-and-formula/

**Screen 3 - positive**

Title: 
`Thank you for your feedback.`

Then:
```
Positive reviews from awesome customers like you help others to feel confident about choosing Mr Beam too. 
Could you take 60 seconds to go to the [Link: Mr Beam Store] and share your happy experiences?

We will be forever grateful. Thank you in advance for helping us out!
Your Mr Beam team
```

 
**Screen 3 - negative**

Title: 
`Thank you for your feedback.`

Then:
```
Did you know:
We have a great online support portal with lots of helpful articles to help out with common problems. There you can also find tips on what materials to use or how to create designs and more. And of course you can reach our friendly support team at any time by writing us a ticket.
[Link: Go to the Mr Beam Support Portal]

Your Mr Beam team
```

khaledsherkawi, 2021-02-12:
@Gallore 
The only thing I kept from before is the back button between screens 1 and 2. The rating bar doesn't look good on top of screen 2. In addition the CSS of the rating bar (on top of screen 2) will be confusing to the user. If we leave it without the their decision highlighted they will think that they have to vote again and if we highlight their decision they will think they cannot change it. With the back button they can choose to go back and answer differently. 

Screen 1:
[image]

Screen 2:
[image]

Screen 3 - positive review:
[image]

Screen 4 - negative review:
[image]

closed 2021-02-12 by Gallore