mrbeam/MrBeamPlugin #1166: Make visible in Left menu which part of the SVG is selected in workspace

clemniem, 2021-02-17:
Use Case: I have an SVG with lots of shapes in it so I use the split, then I am selecting one in the work area, and then I want to know which one it is. The other way around works with making the stroke a bit thicker, so when hovering over the different parts in the left menu. 

[image]

clemniem, 2021-02-17:
I think this might be doable with some light background color changing, like it is often done in menus on hover. Probably a good story also for newbies or rusted frontend devs like me :P

hungerpirat, 2021-04-16:
already fixed in beta.
Thanks for reporting @clemniem

closed 2021-04-16 by hungerpirat