mrbeam/MrBeamPlugin #1190: Plugin crashes if `/etc/mrbeam` not found.

amrsoll, 2021-03-01:
Also the analytics handler throws a fit.

```
 13:26:09 octoprint.plugins.mrbeam.terminal WARNING Can't log to terminal since _printer is None. Message: Can't read device_info_file '/etc/mrbeam' due to exception: [Errno 2] No such file or directory: '/etc/mrbeam'
 13:26:09 octoprint.plugins.mrbeam.util.device_info ERROR Not able to get analytics_handler.
 13:26:09 octoprint.plugins.mrbeam.util.device_info ERROR Could not write exception to analytics, the analytics handler was not initialized.
 13:26:09 octoprint.plugins.mrbeam.util.device_info ERROR Can't read device_info_file '/etc/mrbeam' due to exception: [Errno 2] No such file or directory: '/etc/mrbeam'
```

Gallore, 2021-03-04:
`/etc/mrbeam` is an absolutely required file. I don't expect the plugin to run will without this file. But a well-defined error message would be better, I agree.

amrsoll, 2021-03-04:
I don't understand why. How about generating that file from default values?